lwfinger/rtl8723bs_bt #1: Not Working ? It's it Writen Over ?

harry159821, 2015-06-15:


lwfinger, 2015-06-15:
I just pushed an update and pulled a clean copy. All seems OK for me. Please be specific as to your problem.

harry159821, 2015-06-15:
Sorry for little description, I compiler it and all seem good,just no Bluetooth  devices show ,I see there only once commit,so I guess maybe it's not newest. anywhere,I try to find Android Bluetooth Driver and test it in Linux ,tomorrow. thanks for your reply.

lwfinger, 2015-06-15:
Did you follow the instructions in readme.txt. The BT device in the RTL8723BS is not connected through the USB system. It uses a serial interface. In the startup file, I set the connected device to be /dev/ttyS4. It may be different for your system.

harry159821, 2015-06-16:
So,This Only Connect BT To Serial port /dev/ttyS4, How To Connec Devices? I Kow About Little About That.

lwfinger, 2015-06-16:
The device uses a serial port to create a Bluetooth adapter. Once that is done, it works like any other BT adapter, and you connect to it in the normal manner.

harry159821, 2015-06-16:
But I Haven't See Any Bluetooth Adapter In Top-bar And Settings, How Can I Know It's Running.

lwfinger, 2015-06-16:
Did you check to make certain that /dev/ttyS4 is the correct serial device? That is the correct one for my system, but yours may be different. Do 'dmesg | grep tty' to see the devices that were discovered.

lwfinger, 2015-06-16:
I just modified readme.txt and the start script to have better instructions.

harry159821, 2015-06-19:
After That, I Test Again
1.First, Reinstall Ubuntu In My X80H Pad:

brfore install driver,type  dmesg | grep tty,there already have there ttyS*

ubuntu@ubuntu:~/rtl8723bs_bt-master$ dmesg | grep tty
[    0.000000] console [tty0] enabled
[   11.582110] 00:02: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
[   79.771384] 80860F0A:00: ttyS4 at MMIO 0x90947000 (irq = 39, base_baud = 2764800) is a 16550A
[   79.771866] 80860F0A:01: ttyS5 at MMIO 0x9094d000 (irq = 40, base_baud = 2764800) is a 16550A

2.THEN INSTALL:
ubuntu@ubuntu:~$ cd rtl8723bs_bt-master/
ubuntu@ubuntu:~/rtl8723bs_bt-master$ make
cc -c hciattach_rtk.c
cc -o rtk_hciattach hciattach.c hciattach_rtk.o  
ubuntu@ubuntu:~/rtl8723bs_bt-master$ sudo make install 
mkdir -p /lib/firmware/rtl_bt
cp -p rtlbt_* /lib/firmware/rtl_bt/.
ubuntu@ubuntu:~/rtl8723bs_bt-master$ sudo ./start_bt.sh 
ubuntu@ubuntu:~/rtl8723bs_bt-master$ dmesg | grep tty
[    0.000000] console [tty0] enabled
[   11.582584] 00:02: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
[   80.757668] 80860F0A:00: ttyS4 at MMIO 0x90947000 (irq = 39, base_baud = 2764800) is a 16550A
[   80.758042] 80860F0A:01: ttyS5 at MMIO 0x9094d000 (irq = 40, base_baud = 2764800) is a 16550A

In This Two Section, There Return Nothing When I Type "rfcomm"

ubuntu@ubuntu:~/rtl8723bs_bt-master$ cat /proc/version
Linux version 3.16.0-30-generic (buildd@kissel) (gcc version 4.8.2 (Ubuntu 4.8.2-19ubuntu1) ) #40~14.04.1-Ubuntu SMP Thu Jan 15 17:43:14 UTC 2015
ubuntu@ubuntu:~/rtl8723bs_bt-master$ rfcomm
ubuntu@ubuntu:~/rtl8723bs_bt-master$ 

I Try To Change ttyS4 to ttyS0 in Script, Nothing Change
AND, Bluetooth ICON Not Show In Top-Bar Anytime
I Just What To Know,How Can I Connect Device With This Driver

lwfinger, 2015-06-19:
Your BT device is on ttyS4 - ttyS0 is a regular serial port. You cannot use it.

After you run start_bt.sh, the command 'hcitool dev' should show something. If it does, then 'hcitool scan' should show the BT devices that are broadcasting. If those work, then the driver is working. With my working system, 'rfcomm -a' shows nothing. In addition, my tablet with Ubuntu does show the BT icon in the top bar.

lwfinger, 2015-06-19:
There was a problem in the source. It was looking for the firmware in a different directory than where the Makefile put it. Pull again and retry.

harry159821, 2015-06-21:
I Test It, Can't Work, As I Already Used Do, No error, 'hcitool dev' show nothing,

lwfinger, 2015-06-21:
I changed the start script to log the console output. Please pull again and try to rerun after a reboot. Post the file hciattach.txt at www.pastebin.com, or some other similar location. Then post that link here.

Laszlo-Fiat, 2015-07-04:
I have the same Teclast X80h as the original poster, and It doesn't work for me either. I've just pulled master Today (4 July), and compiled. I have the serial ports in dmesg:

[    1.121694] 80860F0A:00: ttyS1 at MMIO 0x90947000 (irq = 191, base_baud = 2764800) is a 16550A
[    1.122078] 80860F0A:01: ttyS2 at MMIO 0x9094d000 (irq = 192, base_baud = 2764800) is a 16550A

Running the script:  ./start_bt.sh 
Using device /dev/ttyS1 for Bluetooth

Here is the content of hciattach.txt: http://pastebin.com/4LTsKW7k
I don't know if it matters or not, but rfkill doesn't know about the bluetooth device (only the wifi is listed):
 rfkill  list
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no

lwfinger, 2015-07-04:
In hciattach.txt is evidence that the code did not sync with a BT device.  Those lines for ttyS1 and ttyS2 certainly look as if they come from such a device, but who knows.

My script will always pick the first of these two. Just in case you need the second, I have modified it. Please pull again and try starting with with

sudo ./start_bt.sh   ttyS2

That will select the second tty selected.

In addition, please post the output of "uname -r".

Laszlo-Fiat, 2015-07-05:
While looking at my kernel config (based on the stock debian one) I've found that rfkill-gpio was not set. I've added OBDA8723 to rfkill-gpio . and compiled a new kernel (version 4.1.1). With this kernel, the bluetooth initialization WORKED on ttyS1!!!!

I could find nearby devices, Thank you for the driver!

harry159821, 2015-07-06:
congratulate！which debian version U use? Can you show more detail step about your method. I'm new in linux

sigboe, 2015-09-29:
@Laszlo-Fiat Thanks!!! Your suggestion made bluetooth work on my Chuwi Vi8 also.

alssam, 2015-10-16:
Laszlo-Fiat, hi. What does it mean "rfkill-gpio was not set"? If there no line for it in config then do I need to recompile kernel for the driver? If so, then how to add "OBDA8732 to rfkill-gpio"?
Thanks.

harry159821, 2015-10-16:
@sigboe Can You Show How To "added OBDA8723 to rfkill-gpio" and how to complier a new kernel for ubuntu 14

Laszlo-Fiat, 2015-10-17:
@alssam @harry159821 
1. I applied the following patch to the kernel source:
--- a/net/rfkill/rfkill-gpio.c.orig     2015-08-09 16:20:10.673304091 +0200
+++ b/net/rfkill/rfkill-gpio.c  2015-08-09 16:21:24.331672811 +0200
@@ -172,6 +172,7 @@
        { "BCM2E3D", RFKILL_TYPE_BLUETOOTH },
        { "BCM2E40", RFKILL_TYPE_BLUETOOTH },
        { "BCM2E64", RFKILL_TYPE_BLUETOOTH },
+       { "OBDA8723",RFKILL_TYPE_BLUETOOTH },
        { "BCM4752", RFKILL_TYPE_GPS },
        { "LNV4752", RFKILL_TYPE_GPS },
        { },

2. rfkill-gpio was not set to be built in the standard kernel config in Debian:
a@debian:~$ grep RFKILL_GPIO /boot/config-4.1.0-2-amd64 
# CONFIG_RFKILL_GPIO is not set
So I set it to be built as a kernel module with make menuconfig command.
After it was set, the config should have:
CONFIG_RFKILL_GPIO=m
After that, I built the kernel and modules, installed them, and the start_bt.sh script was able to start bluetooth.

lwfinger, 2015-10-17:
At least you now know how to correct the deficiency in the standard Debian kernel configuration.

closed 2015-10-17 by lwfinger