lwfinger/rtl8723bs_bt #2: Bluetooth not working on Onda v975w

hadess, 2015-06-25:
After changing the tty used by hciattach:
```
$ dmesg | grep tty
[    0.000000] console [tty0] enabled
[    1.145803] 00:02: ttyS0 at I/O 0x3f8 (irq = 202, base_baud = 115200) is a 16550A
$ git diff
diff --git a/start_bt.sh b/start_bt.sh
index a1e98a5..2b5f289 100755
--- a/start_bt.sh
+++ b/start_bt.sh
@@ -3,7 +3,7 @@
 # Shell script to install Bluetooth firmware and attach BT part of
 # RTL8723BS
 #
-TTY=/dev/ttyS4
+TTY=/dev/ttyS0
 
 if [ -f /lib/firmware/rtlbt/rtlbt_config ];
 then
$ sudo ./start_bt.sh 
Firmware already loaded
$ hciconfig 
$ cat hciattach.txt 
Realtek Bluetooth init uart with init speed:115200, final_speed:115200, type:HCI UART H5
Realtek Bluetooth :Realtek hciattach version 2.5 

Realtek Bluetooth :3-wire sync pattern resend : 1, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 2, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 3, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 4, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 5, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 6, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 7, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 8, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 9, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 10, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 11, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 12, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 13, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 14, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 15, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 16, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 17, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 18, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 19, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 20, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 21, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 22, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 23, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 24, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 25, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 26, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 27, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 28, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 29, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 30, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 31, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 32, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 33, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 34, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 35, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 36, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 37, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 38, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 39, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 40, len: 8

Realtek Bluetooth ERROR: H5 sync timed out
$ rfkill list
0: OBDA8723:00: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: nfc0: NFC
	Soft blocked: no
	Hard blocked: no
2: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
```

hadess, 2015-06-25:
```
$ grep UART /boot/config-`uname -r`
CONFIG_BT_HCIUART=m
CONFIG_BT_HCIUART_H4=y
CONFIG_BT_HCIUART_BCSP=y
CONFIG_BT_HCIUART_ATH3K=y
CONFIG_BT_HCIUART_LL=y
CONFIG_BT_HCIUART_3WIRE=y
CONFIG_BT_HCIUART_INTEL=y
CONFIG_BT_HCIUART_BCM=y
CONFIG_BT_HCIBTUART=m
CONFIG_SERIAL_8250_NR_UARTS=32
CONFIG_SERIAL_8250_RUNTIME_UARTS=4
# CONFIG_SERIAL_ALTERA_JTAGUART is not set
# CONFIG_SERIAL_ALTERA_UART is not set
# CONFIG_SERIAL_PCH_UART is not set
# CONFIG_SERIAL_XILINX_PS_UART is not set
# CONFIG_SERIAL_FSL_LPUART is not set
CONFIG_SND_MPU401_UART=m
CONFIG_SDIO_UART=m
$ 
```

lwfinger, 2015-06-25:
Device ttyS0 shows up as the emulation of a console TTY. As 'dmesg | grep tty' does not show the registration of any other TTYs, are you sure the BT device uses a serial port? Does it show up with lsusb? If the device is really attached to ttyS0, then you will likely need to change the baud rate as well.

The configuration should be OK. The Ubuntu 3.19 configuration has the ALTERA devices enabled, but none of those drivers are loaded.

hadess, 2015-06-26:
Silly me, Loic already answered my question months ago:
http://www.spinics.net/lists/linux-wireless/msg126863.html

I need the 8250_dw driver compiled and loaded. Building a new kernel now.

hadess, 2015-06-26:
```
$ dmesg | grep tty
[    0.000000] console [tty0] enabled
[    1.142587] 00:02: ttyS0 at I/O 0x3f8 (irq = 202, base_baud = 115200) is a 16550A
[    1.143514] 80860F0A:00: ttyS1 at MMIO 0xa0b4d000 (irq = 191, base_baud = 2764800) is a 16550A
[    1.143901] 80860F0A:01: ttyS2 at MMIO 0xa0b53000 (irq = 192, base_baud = 2764800) is a 16550A

$ git diff
diff --git a/start_bt.sh b/start_bt.sh
index a1e98a5..383d9cb 100755
--- a/start_bt.sh
+++ b/start_bt.sh
@@ -3,7 +3,7 @@
 # Shell script to install Bluetooth firmware and attach BT part of
 # RTL8723BS
 #
-TTY=/dev/ttyS4
+TTY=/dev/ttyS1
 
 if [ -f /lib/firmware/rtlbt/rtlbt_config ];
 then
$ hciconfig 
hci0:	Type: BR/EDR  Bus: UART
	BD Address: 08:D8:34:18:D9:DE  ACL MTU: 1021:8  SCO MTU: 255:16
	UP RUNNING PSCAN ISCAN 
	RX bytes:1822 acl:0 sco:0 events:63 errors:0
	TX bytes:1926 acl:0 sco:0 commands:58 errors:0
```

Yay :)

lwfinger, 2015-06-26:
Congratulations. This gives me an idea on how to make the TTY selection be a part of the start script. Are there any other modules besides 8250_dw that are needed? I'll check the list on my TW100, but having your input would be valuable.

hadess, 2015-06-26:
No other modules needed, no.

lwfinger, 2015-06-26:
I just pushed a script change. It works with no BT device and on the TW100. Does it work on the Onda?

hadess, 2015-07-23:
@garnacho will have to test this when I deliver the tablet to him beginning of August (it's been reset and reinstalled).

dynek, 2015-10-13:
I have just tried it on my onda v975w tablet and it seems to be working OK. I haven't tried further than building it and running the script:

[31260.098917] Bluetooth: Core ver 2.20
[31260.108277] NET: Registered protocol family 31
[31260.115764] Bluetooth: HCI device and connection manager initialized
[31260.122967] Bluetooth: HCI socket layer initialized
[31260.130144] Bluetooth: L2CAP socket layer initialized
[31260.137349] Bluetooth: SCO socket layer initialized
[31260.148457] Bluetooth: HCI UART driver ver 2.3
[31260.156113] Bluetooth: HCI UART protocol H4 registered
[31260.163284] Bluetooth: HCI UART protocol BCSP registered
[31260.170417] Bluetooth: HCI UART protocol LL registered
[31260.177509] Bluetooth: HCI UART protocol ATH3K registered
[31260.184562] Bluetooth: HCI UART protocol Three-wire (H5) registered
[31260.191605] Bluetooth: HCI UART protocol BCM registered
[31260.206627] Bluetooth: Out-of-order packet arrived (7 != 0)
[31260.584730] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[31260.592004] Bluetooth: BNEP filters: protocol multicast
[31260.599245] Bluetooth: BNEP socket layer initialized
[31292.404675] Bluetooth: hci0 sending frame failed (-16) <-- when I killed the process

braiam, 2017-02-17:
@hadess did you test on kernel 4.9? I have the exact same card (the wifi works fine btw, thanks!) but neither hciconfig nor rfkill list the card. It prints the dmesg message, but otherwise it prints the H5 sync timed out error. I compiled the kernel using Debian .config and patches needed for the wifi card, is there anything special I have to do?

hadess, 2017-02-18:
I don't have this hardware any more, and my other BayTrail tablet can't install my preferred distribution. Read the above carefully and make sure you have the expected device drivers compiled.