lwfinger/rtl8723bs_bt #4: Multiple tty devices, none working

pgquiles, 2015-11-09:
Platform: Pipo X7 mini PC
Software: Ubuntu 15.10 with kernel 4.3.0-0.6

On my Pipo X7 miniPC, two serial devices are found:

$ dmesg | grep tty
[    0.000000] console [tty0] enabled
[   13.338680] 00:02: ttyS0 at I/O 0x3f8 (irq = 199, base_baud = 115200) is a 16550A
[   35.740916] systemd[1]: [/lib/systemd/system/casper.service:10] Failed to parse input specifier, ignoring: force-tty
[   35.769520] systemd[1]: Created slice system-getty.slice.
[   36.602815] 80860F0A:00: ttyS4 at MMIO 0x90947000 (irq = 191, base_baud = 2764800) is a 16550A
[   36.603854] 80860F0A:01: ttyS5 at MMIO 0x9094d000 (irq = 192, base_baud = 2764800) is a 16550A

I have tried both of them but none works.

With serial speed 115200:

# ./rtk_hciattach -n -s 115200 /dev/ttyS4 rtk_h5
Realtek Bluetooth init uart with init speed:115200, final_speed:115200, type:HCI UART H5                                                                                                                                                                                       
Realtek Bluetooth :Realtek hciattach version 2.5                                                                                                                                                                                                                               
                                                                                                                                                                                                                                                                               
Realtek Bluetooth :3-wire sync pattern resend : 1, len: 8                                                                                                                                                                                                                      
                                                                                                                                                                                                                                                                               
Realtek Bluetooth :Get SYNC Resp Pkt                                                                                                                                                                                                                                           
                                                                                                                                                                                                                                                                               
Realtek Bluetooth :Get SYNC pkt-active mode                                                                                                                                                                                                                                    
                                                                                                                                                                                                                                                                               
Realtek Bluetooth :3-wire config pattern resend : 1 , len: 10                                                                                                                                                                                                                  
Realtek Bluetooth :Get CONFG pkt-active mode                                                                                                                                                                                                                                   
                                                                                                                                                                                                                                                                               
Realtek Bluetooth :Get CONFG resp pkt-active mode                                                                                                                                                                                                                              
                                                                                                                                                                                                                                                                               
Realtek Bluetooth :H5 init finished                                                                                                                                                                                                                                            
                                                                                                                                                                                                                                                                               
Realtek Bluetooth :config offset(f4),length(8)                                                                                                                                                                                                                                 
Realtek Bluetooth :config baud rate to :4928002, hwflowcontrol:5f, 1                                                                                                                                                                                                           
Realtek Bluetooth :config offset(27),length(1)                                                                                                                                                                                                                                 
Realtek Bluetooth :config offset(fe),length(1)                                                                                                                                                                                                                                 
Realtek Bluetooth :config offset(15b),length(4)                                                                                                                                                                                                                                
Realtek Bluetooth :config offset(1e3),length(1)                                                                                                                                                                                                                                
Realtek Bluetooth :Get config baud rate from config file:4928002                                                                                                                                                                                                               
Realtek Bluetooth ERROR: Can't access firmware, errno:2                                                                                                                                                                                                                        
Realtek Bluetooth ERROR: Get BT firmware error                                                                                                                                                                                                                                 
Can't initialize device: No such file or directory                                                                                                                                                                                                                             
root@kubuntu:/cdrom/rtl8723bs_bt# ./rtk_hciattach -n -s 115200 /dev/ttyS4 rtk_h5                                                                                                                                                                                               
Realtek Bluetooth init uart with init speed:115200, final_speed:115200, type:HCI UART H5                                                                                                                                                                                       
[... cut ...]
Realtek Bluetooth :3-wire sync pattern resend : 33, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 34, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 35, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 36, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 37, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 38, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 39, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 40, len: 8

Realtek Bluetooth ERROR: H5 sync timed out

pgquiles, 2015-11-09:
$ sudo ./rtk_hciattach -n -s 115200 /dev/ttyS4 rtk_h5
Realtek Bluetooth init uart with init speed:115200, final_speed:115200, type:HCI UART H5
Realtek Bluetooth :Realtek hciattach version 2.5 

Realtek Bluetooth :3-wire sync pattern resend : 1, len: 8

Realtek Bluetooth :Get SYNC Resp Pkt

Realtek Bluetooth :Get SYNC pkt-active mode

Realtek Bluetooth :3-wire config pattern resend : 1 , len: 10
Realtek Bluetooth :Get CONFG pkt-active mode

Realtek Bluetooth :Get CONFG resp pkt-active mode

Realtek Bluetooth :H5 init finished

Realtek Bluetooth :config offset(f4),length(8)
Realtek Bluetooth :config baud rate to :4928002, hwflowcontrol:5f, 1
Realtek Bluetooth :config offset(27),length(1)
Realtek Bluetooth :config offset(fe),length(1)
Realtek Bluetooth :config offset(15b),length(4)
Realtek Bluetooth :config offset(1e3),length(1)
Realtek Bluetooth :Get config baud rate from config file:4928002
Realtek Bluetooth :Load FW OK
Realtek Bluetooth :RTK send HCI_VENDOR_READ_RTK_ROM_VERISION_Command

Realtek Bluetooth :Received reliable seqno 0 from card
Realtek Bluetooth :receive hci command complete event with command:1001

Realtek Bluetooth :Read RTK LMP version with Status:0
Realtek Bluetooth :gLmpVersion = 0x8723
Realtek Bluetooth :RTK send HCI_VENDOR_READ_RTK_ROM_VERISION_Command

Realtek Bluetooth :Received reliable seqno 1 from card
Realtek Bluetooth :receive hci command complete event with command:fc6d

Realtek Bluetooth :Read RTK rom version with Status:0
Realtek Bluetooth :rtk_hw_cfg.eversion = 1
Realtek Bluetooth :rtk_get_fw_project_id: opcode 0, len 1, data 1
Realtek Bluetooth :fw_ver 0x1e3ee40e, patch_num 2
Realtek Bluetooth :patch length is 0x5e90
Realtek Bluetooth :start offset is 0x4f00
Realtek Bluetooth :fw: exists, config file: exists
Realtek Bluetooth :baudrate in change speed command: 0x2 0x80 0x92 0x4 

Realtek Bluetooth :Received reliable seqno 2 from card
Realtek Bluetooth :receive hci command complete event with command:fc17

Realtek Bluetooth :Change BD Rate with status:0
Realtek Bluetooth :final_speed 1500000

Realtek Bluetooth :hw flow control enable
Realtek Bluetooth :iEndIndex:96  iLastPacketLen:71 iAdditionpkt:4

Realtek Bluetooth :hci_download_patch tx_index:0 rx_index: -1

Realtek Bluetooth :Received reliable seqno 3 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 0

Realtek Bluetooth :hci_download_patch tx_index:1 rx_index: 0

Realtek Bluetooth :Received reliable seqno 4 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 1

Realtek Bluetooth :hci_download_patch tx_index:2 rx_index: 1

Realtek Bluetooth :Received reliable seqno 5 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 2

[... 469 lines cut ...]

Realtek Bluetooth :hci_download_patch tx_index:97 rx_index: 96

Realtek Bluetooth :Received reliable seqno 4 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 97

Realtek Bluetooth :hci_download_patch tx_index:98 rx_index: 97

Realtek Bluetooth :Received reliable seqno 5 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 98

Realtek Bluetooth :hci_download_patch tx_index:99 rx_index: 98

Realtek Bluetooth :Received reliable seqno 6 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 99

Realtek Bluetooth :Send FW last command
Realtek Bluetooth :hci_download_patch tx_index:100 rx_index: 99

Realtek Bluetooth :Received reliable seqno 7 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 100

Realtek Bluetooth :Init Process finished
Realtek Bluetooth post process
Device setup complete

lwfinger, 2015-11-09:
Before you dump a bunch of output onto the site, please read it. One of those sections says that the firmware was not available.

Did you 'sudo make install'?

pgquiles, 2015-11-09:
Sorry about the long output, the Pipo X7 is very unstable due to lack of refrigeration and hangs all the time, so I was posting in a hurry. (I have procured fans and thermal paste now :-) )

I cannot remember if I did sudo make install, probably not because I thought the firmware from linux-firmware.git would work. Where did you get the rtl_fw firmware file from? It seems it's neither one of the files available in linux-firmware.git :-?

Unrelated to this: I am packaging this driver for Debian, I will send you a pull request with a couple of changes later.

lwfinger, 2015-11-09:
Firmware in linux-firmware.git is ONLY for drivers that are in the kernel. As this one is not, you may draw the conclusion. The firmware came from Realtek. They provide no firmware source, but it is licensed for redistribution.