lwfinger/rtl8723bs_bt #6: can't initialize device on two different machines

daduke, 2015-12-04:
Hi,

I have two Baytrail machines:

  * Medion Akoya S2217 (MD 99512): Atom Z3735F, /sys/bus/sdio/devices/mmc1:0001:1/device: 0xb723
    * 80860F0A:00: ttyS4 at MMIO 0x9093b000 (irq = 191, base_baud = 2764800)
is a 16550A
    * 80860F0A:01: ttyS5 at MMIO 0x90941000 (irq = 192,
base_baud = 2764800) is a 16550A
  * Lenovo Yoga Tablet 2 1051F: Atom Z3745, /sys/bus/sdio/devices/mmc1:0001:1/device: 0x4324, 0001:2/device: 0x4324
    * 80860F0A:00: ttyS4 at MMIO 0x9091d000 (irq
= 191, base_baud = 2764800) is a 16550A
    * 80860F0A:01: ttyS5 at MMIO 0x9091f000 (irq
= 193, base_baud = 2764800) is a 16550A

I run a homebrew 4.2.4 kernel with the SDIO patches compiled in. On both machines I can't initialize the BT device:

    Realtek Bluetooth init uart with init speed:115200, final_speed:115200, type:HCI UART H5
    Realtek Bluetooth :Realtek hciattach version 2.5
    Realtek Bluetooth :3-wire sync pattern resend : 1, len: 8

just counts up to 40 and stops. I tried with both ttyS4 and S5 and 115200 as well as 2764800 baud.

what else can I do to get this up and running?

thanks a lot,
-Christian

lwfinger, 2015-12-04:
As with the wifi device, there was nothing in the tables to recognize this device. I just pushed a change that should fix that. Please try.

daduke, 2015-12-05:
Hi Larry,

no change I'm afraid. Still won't recognize. What can I do to help debug?

thanks,
-Christian

daduke, 2015-12-07:
Hi Larry,

after we figured out that the Lenovo uses a Broadcom chip (and I got it kinda working with the correct driver) I still have the bluetooth problem on the Medion: vendor/device 0x024c/0xb723 points to an RTL8723B so I guess it should be working. But I have the same problem: hciattach just counts to 40, no connection.

thanks,
-Christian