lwfinger/rtl8723bs_bt #7: Onda V919 3G Dual Os 64Gb : H5 sync timed out

jon1012, 2015-12-21:
Hi,

I tried all the ttyS* on my device and none works.
Wifi is working perfectly (recompiled kernel with fixes from the patch/ directory of rtl8723bs repo), but no way to get the bluetooth work.

Command outputs : 

======================
$ dmesg | grep tty
[    0.000000] console [tty0] enabled
[    1.672313] 00:02: ttyS0 at I/O 0x3f8 (irq = 199, base_baud = 115200) is a 16550A
[    6.740893] systemd[1]: Created slice system-getty.slice.
[    7.103121] 80860F0A:00: ttyS4 at MMIO 0x90913000 (irq = 191, base_baud = 2764800) is a 16550A
[    7.103517] 80860F0A:01: ttyS5 at MMIO 0x90919000 (irq = 192, base_baud = 2764800) is a 16550A

======================

hciattach.txt: 

Realtek Bluetooth init uart with init speed:115200, final_speed:115200, type:HCI UART H5
Realtek Bluetooth :Realtek hciattach version 2.5 
Realtek Bluetooth :3-wire sync pattern resend : 1, len: 8
Realtek Bluetooth :3-wire sync pattern resend : 2, len: 8
Realtek Bluetooth :3-wire sync pattern resend : 3, len: 8

[... cut for brievety]

Realtek Bluetooth :3-wire sync pattern resend : 40, len: 8
Realtek Bluetooth ERROR: H5 sync timed out

=====================

I tried to use it on all the tty (not just ttyS4), change the baudrate, even ried rtk_h4 with no luck.

Any idea why it might happen ?

Thanks,
Jonathan

jon1012, 2016-01-08:
no idea ? :(

lwfinger, 2016-01-08:
I do not. On the TW100 that I have, BT just worked. You might check the output of 'dmesg | grep tty' to see what tty's are created.

jon1012, 2016-01-08:
I just pasted the output of that command on the ticket, ttyS4 and ttyS5 are created. But the bluetooth module doesn't answer on any of them. 

I am running the sdio patched kernel with the rtl8723bs module for wifi.

lwfinger, 2016-01-08:
I would think that ttyS4 is the correct unit. From the dmesg output, a baud rate of 2764800 would be suggested; however, that is a strange number. I would also try 115200, 230400, 460800, 921600, and 1843200.

jon1012, 2016-01-08:
Thanks for your insights.
I just tried those values with no luck, I still have this H5 timeout error.

I used this command with all the values you suggested :
sudo ./rtk_hciattach -n -s <baudrate> /dev/ttyS4 rtk_h5

lwfinger, 2016-01-08:
Does the hardware work with Windows?

jon1012, 2016-01-10:
Didn't have a windows on the tablet to check anymore, so it took a bit of time.
Yes, it works under windows and android.

lwfinger, 2016-01-11:
Do you know where the source is for the version of Android that you checked?

jon1012, 2016-01-11:
I didn't find a source code sorry :(
An image is here : https://mega.nz/#!PA1nxaAK!NpTtZVaCQadeqa437K3mUsSf9OEN7jDMshmf5hI8wOk

TobleMiner, 2016-01-31:
I'm having the exact same issue on a TrekStor SurtTab wintron 7.0. The kernel reports this:
```
[   18.583782] 80860F0A:00: ttyS1 at MMIO 0x48919000 (irq = 191, base_baud = 2764800) is a 16550A
[   18.588570] 80860F0A:01: ttyS2 at MMIO 0x4891b000 (irq = 192, base_baud = 2764800) is a 16550A
```

But unluckily the H5 sync times out on both ttyS1 and ttyS2. When the sync pattern is sent no data can be recieved at the stty.  I'm totally stuck because I have no idea how to tackle this problem. You seem to have quite some insight in the operation of realtek chipsets. Do you have any hints for further debugging? Any help is greatly appreciated.

lwfinger, 2016-01-31:
I know very little about the BT devices; however, most of the Realtek serial BT devices are not connected to ttyS1 or ttyS2. These devices are usually real serial ports even though there is no external connector for them. Every Realtek BT device that I know about uses ttyS4.

TobleMiner, 2016-01-31:
Hm, ok. Thanks for the info. Bluetooth somewhat works under Windows so there must be some serial connection. Maybe my tablet uses some strange seial interface that's not recognized by Linux? There are only ttyS0-2. I'm quite puzzled about that.

jasonabele, 2016-02-01:
I am not at all familiar with the Onda devices, but on the NextThingCo CHIP boards, the rtl8723bs bluetooth device is wired to the second UART of the SoC and appears at /dev/ttyS1 ... the attachment to something like ttyS[0-4] is specific to how the Onda device is physically constructed.

One thing that may help get people going a little better is a small observation that I made while getting the rtl8723bs bluetooth working on CHIP was:

The bluetooth firmware is downloaded to the rtl8723bs at 115200bps by the rtk_hciattach command.  After firmware download, the rtk_hciattach command sends an H5 command to change the UART rate to something higher than 115200bps.

However, the rtk_hciattach command does nothing to probe the UART before attempting to send the bluetooth firmware at 11520bps.

This leads to a strange fault where the second attempt to run an rtk_hciattach will fail because the firmware has already been downloaded an the rtl8723bs device is set to a higher communication rate.  Warm reboots can also exhibit this same issue as the rtl8723bs may still be configured for the higher bitrate and thus the new attempt to download firmware at 115200bps will fail and thus all of the hciattach fails.

At NextThing, we are carrying a hack for now that toggles the reset line of the rtl8723bs before calling rtk_hciattach, but this would be hard to do on other devices without knowledge of how the rtl8723bs is wired to the SoC.

I am working on a better solution where the firmware download is skipped if the UART is detected to be in a higher bitrate and I will try to remember to ping this issue/repo when we have a better fix.

Hope that helps someone fix this up for other rtl8723bs devices.

pcercuei, 2016-03-09:
Same issue on a Teclast x89.

youling257, 2016-03-30:
onda v891w ，same problem

BotCyborg, 2016-06-04:
I had identical issue on a no-name baytrail device in laptop form factor from alibaba.
Applied the patches from [here](https://github.com/hadess/rtl8723bs) and then installed [this](https://github.com/hadess/rtl8723bs) and [this](https://github.com/lwfinger/rtl8723bs_bt). Was getting time outs after 40 counts like @jon1012 .

The following resolved it -->

I added the lines from [here](https://raw.githubusercontent.com/Manouchehri/vi8/master/Ubuntu_support_files/rfkill.patch). (had to edit by hand, patch didn't work, since there are other changes on 4.5 kernel). Then enabled CONFIG_RFKILL_GPIO=m in kernel config and recompiled kernel. Re-installed rtl8723bs and rtl8723bs_bt and it worked.

Hope it helps someone else.

pcercuei, 2016-06-07:
That was it! Bluetooth is now working here. Thanks!

MalakiLab, 2016-06-08:
Adding { "OBDA8723", RFKILL_TYPE_BLUETOOTH }, in rfkill-gpio.c did the trick for me too.

jon1012, 2016-06-08:
Wow, nice !

I'll try it, maybe that tablet won't sit in a drawer anymore :)

lwfinger, 2016-06-08:
Even though adding that line works, that is NOT the long-term solution for the kernel. The RFKILL_TYPE_BLUETOOTH entries from that table have been moved. I have tried to find the "correct and approved" way to make this change, but no luck so far. If anyone out there has any suggestions, then please post them.

MalakiLab, 2016-06-08:
To what i see in the in the .c, it wasn't moved, it was replaced by an automatic detection based on the ACPI properties of the device. Not completely sure though.

As of kernel 3.19, the kernel use the Unified Device Properties Interface. http://events.linuxfoundation.org/sites/events/files/slides/unified_properties_API_0.pdf

I'm pretty sure the ACPI table don't contain the type of device it is, and is not recognized as RFKILL_TYPE_BLUETOOTH, and skipped. Those still in the hardcoded table seems like similar exceptions.

I can be wrong. I stopped following and reading kernel sources when they moved to 3.x, now i will be honest, i am lost in new implementations.

MatejKovacic, 2016-06-27:
I have PCG02U Ubuntu Stick PC (bought from AliExpress), running Ubuntu 14.04.4 LTS and kernel 3.16.0-71.

I run `sudo ~/rtl8723bs_bt/start_bt.sh`, but I got this error `H5 sync timed out`. My problem is described here:
https://github.com/lwfinger/rtl8723bs_bt/issues/13

Any idea how to solve this? @BotCyborg, how exactly did you solve the problem?

protectivedad, 2016-06-28:
Thanks for the solution. Just for anyone who also has this problem. I got the bluetooth working with the patch described here with RFKILL_TYPE_BLUETOOTH. I'm using HP Stream 7 and the ACPI identification is ODBA8723 (rtl8923bs). I also noticed this ACPI being discussed in hadess work on the rtl8723bs driver

kahrloz, 2016-07-07:
where exactly did you add those lines?

BotCyborg, 2016-07-07:
@kahrloz add it in the in the rfkill-gpio.c file, in the linux kernel source and then recompile the kernel.
(P.S. It does the job, but is only an ugly workaround)

@MatejKovacic I have already mentioned the 4 things that I did, in my last post. I'm afraid you would have to ask google if you need detailed help with applying patches and recompiling the kernel. My apologies, but describing everything here would be beyond the scope of this thread.

kahrloz, 2016-07-08:
i looked for that file, but couldn't find it

kahrloz, 2016-07-10:
@BotCyborg where is that file located? Can't find it

BotCyborg, 2016-07-10:
@kahrloz Hi! Here it is, in the kernel source
https://github.com/torvalds/linux/blob/master/net/rfkill/rfkill-gpio.c
Hope that helps.

kahrloz, 2016-07-10:
@BotCyborg  i found a rfkill-gpio.h file, is that it?

BotCyborg, 2016-07-10:
@kahrloz You should be able to see these files in the net/rfkill/ directory as shown here,
http://lxr.free-electrons.com/source/net/rfkill/?v=4.4
Where did you get your kernel source from?

kahrloz, 2016-07-10:
@BotCyborg  what if those files are not there?

BotCyborg, 2016-07-10:
@kahrloz Then most likely you are looking at the wrong place or using some custom kernel source. Where did you download the kernel source from?

Also are you very sure you are not looking at the linux-header sources or the include directory?

pietrushnic, 2016-08-13:
@lwfinger it looks like I have very similar problem with Allwinner A20 custom board. I was able to integrated driver with buildroot and build it with 4.7 kernel. I'm sure that I'm using correct `ttySX` device based on MMIO. Unfortunately I'm getting;

```
Realtek Bluetooth ERROR: H5 sync timed out
```

`BT_DIS` was verified to be high. Do I need any additional code to drive `BT_WAKE` ? I also assume GPIO to which I have `BT_HOST_WAKE` connected should be configured as input, but does driver or any kernel will know about that without correct device tree ?

@jasonabele if I read CHIP schematics correctly `BT_WAKE` is connected directly to AXP209, do you know how this pin should be driven ? Also `BT_HOST_WAKE` is `PG0` and I can't see `PG0` configuration in CHIP device tree in mainline.

BTW WiFi works to some extent, so I assume hardware is connected correctly in this area.

pietrushnic, 2016-08-14:
@lwfinger @jasonabele I managed to make some progress by toggling `BT_DIS` before starting to send anything on UART. In that way it looks like I'm able to finish firmware upload:

```
Realtek Bluetooth :Send FW last command
Realtek Bluetooth :hci_download_patch tx_index:100 rx_index: 99

Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:0
Realtek Bluetooth ERROR: patch timerout, retry:
```

Full `dmesg` can be found [here](http://paste.ubuntu.com/23056918/). I'm using recent `hadess/rtl8723bs` if this help anything.

@hadess do you have any insights about BT part of this chip ?

jingber, 2016-08-15:
Same problem here, even with OBDA patch.  I have ttyS1 and ttyS2, neither work.

H5 sync time out

lwfinger, 2016-08-15:
Devices ttyS1 and ttys2 are real serial ports, not the one created by the Bluetooth device. If you are not seeing other ttySx devices, then your BT device is not being found. I have no idea what errors there might be in the kernel configuration, but I would suspect that as the problem.

pietrushnic, 2016-08-16:
@jingber OBDA patch may work only on platforms with ACPI (mostly Intel platforms).

@lwfinger have you got any thoughts about `patch timerout` problem ?

lwfinger, 2016-08-16:
No. I have no hardware that shows this problem, and I have no plans to buy anything new.

madivak, 2017-02-17:
@pietrushnic i got the BT device to wake up and scanned for available bluetooth signals. Am using Allwinner A20 and the RTL8723bs on a custorm board.

```
hciconfig -a
hci0:   Type: BR/EDR  Bus: UART
        BD Address: 8D:18:D9:35:7C:3C  ACL MTU: 1021:8  SCO MTU: 255:16
        UP RUNNING 
        RX bytes:5737 acl:0 sco:0 events:94 errors:0
        TX bytes:1971 acl:0 sco:0 commands:75 errors:0
        Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
        Link policy: RSWITCH HOLD SNIFF PARK 
        Link mode: SLAVE ACCEPT 
        Name: 'RTK_BT_4.0'
        Class: 0x000000
        Service Classes: Unspecified
        Device Class: Miscellaneous, 
        HCI Version: 4.0 (0x6)  Revision: 0x1e3e
        LMP Version: 4.0 (0x6)  Subversion: 0xe40e
        Manufacturer: Realtek Semiconductor Corporation (93)

hcitool dev
Devices:
        hci0    8D:18:D9:35:7C:3C
hcitool scan
Scanning ...
        54:44:08:30:FD:06       x2:
        74:04:2B:4E:4D:31       Lenovo K3 Note
```

pietrushnic, 2017-03-12:
@madivak according to our email discussion you managed to solve that issues. It would be great if you can leave here a note for community benefit.

madivak, 2017-03-13:
@pietrushnic sorry about that. 
After cross-compiling and ensuring all files available at /lib/firmware/rtl_bt (rtlbt_config and rtlbt_fw)
i created 2 scripts to initialize BT. I didnt use the *start_bt.sh*

1st script is to initialize BT_WAKE and BT_WAKE_HOST, here is the script, run this after boot.
```
#!/bin/bash
#initialize pin BT_WAKE_HOST (on GPIO239 on my board) to register as gpio_in with rising edge
echo "239"  > /sys/class/gpio/export   
echo "in"  > /sys/class/gpio/gpio239/direction  
echo "rising"  > /sys/class/gpio/gpio239/edge

#initialize pin BT_WAKE (on GPIO238 on my board) to register as gpio_out and conf HIGH
echo "238" > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio238/direction
echo "1" > /sys/class/gpio/gpio238/value

```
2nd script runs the BT driver
```
#!/bin/bash
#Run Bluetooth driver (BT attached to ttyS2 on my board)

rtk_hciattach -n -b -s 115200 /dev/ttyS2 rtk_h5 > hciattach.txt 2>&1 &
/bin/sleep 5		#allow hci0 interface to initialise (u can spot the hci0 when u check #ps)
hciconfig               #show if hci inteface is up
hciconfig hci0 up       #bring up the hci inteface
hciconfig -a            #show if hci inteface is up
hciconfig hci0 piscan   #Make my device discoverable
hcitool scan            #scan available BT signals
```

So after this, i culd now use the Bluetoothctl to do all Bluetooth functionalities.

madivak, 2017-04-06:
@pietrushnic I managed to connect and send files via Bluetooth using rfcomm. Will give u much in-site in a few when I get to the office.

xchetah, 2017-05-02:
@madivak  hi , I have a23 tablet I followed your steps and got bluetooth to work , but after a while all commands return timeout : 
hci0 command tx timeout 
and strange error : out-of-order packet arrived , got 7 expected 0
any help is apreciated

madivak, 2017-05-03:
@xchetah that's strange. I have never encountered the "timeout" once i scripted the 2 Scripts listed above. If your issue involves scanning and u r using "Bluetoothctl" you can try and leave the "scan on/off" option "ON" and see whether there is timeout. You can go ahead and remove the "hcitool scan" from the script and do it withing "Bluetoothctl". Am not sure whether that will solve your problem.

xchetah, 2017-05-03:
@madivak thanks for your reply , anyway that didn't fix the problem I think it's either rtk_hciattach or tty problem , the problem happens if i delay any command , for example if I delay hci0 up after hciattach , it wont bring the interface up , so all i have to do is to keep the device busy with commands which is not practical , btw I'm using allwinner kernel 3.4 , got problems with mainline kernel

madivak, 2017-05-03:
@xchetah have u tried re-compiling it. Were the mainline issues linked with compiling this driver? Because that wasn't my case. Either try recompiling the kernel and the driver and see what happens.

xchetah, 2017-05-03:
@madivak in mainline kernel rtk_hciattach fails with : 
peer acked invalid packet
ERROR : h5 hdr checksum error
rtk get lmp version timeout
and so on

madivak, 2017-05-03:
yikes! which kernel version is that?
I can give you my kernel.config for 4.3.2 you download the Kernel source and recompile both and see what happens. That will be my best effort to ur problem, since i made it work and moved on to other minor-projects.
 Here is a link to configuration for Buildroot, kernel and my dts file
If u have the time you can build your system, but make amends to ur A23 specification. especially change a few things in buildroot as well. tell me how it goes.
LINK: [http://linux.4shared.com](url)

xchetah, 2017-05-03:
@madivak main kernel 4.11-rc1 works now , i missed a regulator enable in the dts file , works perfectly thanks for your effort

madivak, 2017-05-03:
The mmc section?
OK, Glad i could be of help. Cheers. :100: :+1:

xchetah, 2017-05-03:
@madivak 
this is the parameters from my script.bin 
[wifi_para]
wifi_used = 1
wifi_sdc_id = 1
wifi_usbc_id = 1
wifi_usbc_type = 1
wifi_mod_sel = 9
wifi_power = "axp22_dldo1"
wifi_power_ext1 = "axp22_dldo2"
wifi_power_ext2 = ""
wifi_power_switch = port:power0<1><0><default><0>
rtl8723bs_vq0_chip_en = port:PL11<1><default><default><0>
rtl8723bs_vq0_wl_regon = port:PL06<1><default><default><0>
rtl8723bs_vq0_wl_host_wake = port:PL07<4><default><default><0>
rtl8723bs_vq0_bt_regon = port:PL08<1><default><default><0>
rtl8723bs_vq0_bt_wake = port:PL10<1><default><default><0>
rtl8723bs_vq0_bt_host_wake = port:PL09<4><default><default><0>
rtl8723bs_vq0_lpo_use_apclk = 0

[bt_para]
bt_used = 1
bt_uart_id = 1

I didnt enable wifi_power_ext1 = "axp22_dldo2" in the dtsi , which resulted in the errors mentioned earlier

emelcher, 2017-07-17:
With kernel 4.8 Bluetoooth worked with this patch:
https://github.com/burzumishi/linux-baytrail-flexx10/tree/master/kernel/patches/v4.8
In kernel 4.12, rtl8723bs wifi got included. However, even with that patch applied, the "base_baud = 2764800" appears in dmesg and 'ERROR: H5 sync timed out" appears in hciattach.txt.
This is my computer:
http://lad.dsc.ufcg.edu.br/lad/pmwiki.php?n=Lad.Tablet
Is there a way to get Bluetooth working using kernel 4.12 ?

madivak, 2017-07-17:
With that issue I suspect its something to do with your DTS file. Check previous comments above and see guys had the same issue. Make sure your DTS file is properly configured. Check my comment above u will find a link for my DTS file and kernel. Properly look at the DTS file.

emelcher, 2017-07-17:
I forgot to enable CONFIG_RFKILL_GPIO in the kernel .config, Sorry for the silly mistake.

youyudedasan, 2017-10-03:
I drive rtl8723ds with rtk_h5 protocol on ARM 9,kernel 3.10.65,bluez5.38.
when i ran hciattach -n -s 115200 ttyS2 rtk_h5,I got logs below:
`root@TinaLinux:/# hciattach -n -s 115200 ttyS2 rtk_h5
Realtek Bluetooth init uart with init speed:115200, final_speed:115200, type:HCI UART H5
Realtek Bluetooth :Realtek hciattach version 3.1

Realtek Bluetooth :3-wire sync pattern resend : 1, len: 8

Realtek Bluetooth :Get SYNC Resp Pkt

Realtek Bluetooth :Get SYNC pkt-active mode

Realtek Bluetooth :3-wire config pattern resend : 1 , len: 10
Realtek Bluetooth :Get CONFG pkt-active mode

Realtek Bluetooth :Get CONFG resp pkt-active mode

Realtek Bluetooth :H5 init finished

Realtek Bluetooth :RTK send HCI_VENDOR_READ_RTK_ROM_VERISION_Command

Realtek Bluetooth :Received reliable seqno 0 from card
Realtek Bluetooth :receive hci command complete event with command:1001

Realtek Bluetooth :Read Local Version Information with Status:0
Realtek Bluetooth :HCI Version 0x08
Realtek Bluetooth :HCI Revision 0x000d
Realtek Bluetooth :LMP Subversion 0x8723
Realtek Bluetooth :RTK send HCI_VENDOR_READ_RTK_ROM_VERISION_Command

Realtek Bluetooth :Received reliable seqno 1 from card
Realtek Bluetooth :receive hci command complete event with command:fc6d

Realtek Bluetooth :Read RTK rom version with Status:0
Realtek Bluetooth :LMP Subversion 0x8723
Realtek Bluetooth :EVersion 2
Realtek Bluetooth :IC: RTL8723DS

Realtek Bluetooth :Firmware/config: rtl8723d_fw, rtl8723d_config

Realtek Bluetooth :config offset(00f4),length(01)
Realtek Bluetooth :config offset(00f6),length(02)
Realtek Bluetooth :config offset(00fa),length(02)
Realtek Bluetooth :config baud rate to :4928002, hwflowcontrol:5f, 1
Realtek Bluetooth :config offset(00d9),length(01)
Realtek Bluetooth :config offset(00e4),length(01)
Realtek Bluetooth :Get config baud rate from config file:252c00a
Realtek Bluetooth :Load FW OK
Realtek Bluetooth :rtk_get_fw_project_id: opcode 0, len 1, data 9
Realtek Bluetooth :fw_ver 0xaa5d675f, patch_num 3
Realtek Bluetooth :chip id 0x0001
Realtek Bluetooth :chip id 0x0002
Realtek Bluetooth :chip id 0x0003
Realtek Bluetooth :patch length is 0x7594
Realtek Bluetooth :start offset is 0x4980
Realtek Bluetooth :Svn version:    18994

Realtek Bluetooth :Coexistence: BTCOEX_20161108-1010

Realtek Bluetooth :fw: exists, config file: exists
Realtek Bluetooth :Total len 30147 for fw/config
Realtek Bluetooth :baudrate in change speed command: 0xa 0xc0 0x52 0x2

Realtek Bluetooth :Received reliable seqno 2 from card
Realtek Bluetooth :receive hci command complete event with command:fc17

Realtek Bluetooth :Change BD Rate with status:0
Realtek Bluetooth :final_speed 230400

Realtek Bluetooth :hw flow control enable
[... 617 lines cut ...]
Realtek Bluetooth :rtk_hw_cfg.rx_index 121

Realtek Bluetooth :Send additional packet 122
Realtek Bluetooth :hci_download_patch tx_index:122 rx_index: 121

Realtek Bluetooth :Received reliable seqno 5 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 122

Realtek Bluetooth :Send additional packet 123
Realtek Bluetooth :hci_download_patch tx_index:123 rx_index: 122

Realtek Bluetooth :Received reliable seqno 6 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 123

Realtek Bluetooth :Send end packet 252
Realtek Bluetooth :Send FW last command
Realtek Bluetooth :hci_download_patch tx_index:124 rx_index: 123

Realtek Bluetooth ERROR: h5 hdr checksum error!!!
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:0
Alarm clock`

PS:i show the uart speed to 115200.i set alarm timer to 10s in case of alarming on downloading patch

dmarkey, 2017-11-26:
```
root@amlogic:~/rtl8723bs_bt# ./rtk_hciattach -n -s 115200 /dev/ttyS1 rtk_h5
Realtek Bluetooth init uart with init speed:115200, final_speed:115200, type:HCI UART H5
Realtek Bluetooth :Realtek hciattach version 2.5 

Realtek Bluetooth :3-wire sync pattern resend : 1, len: 8

Realtek Bluetooth :Get SYNC Resp Pkt

Realtek Bluetooth :3-wire config pattern resend : 1 , len: 10
Realtek Bluetooth :Get SYNC pkt-active mode

Realtek Bluetooth :Get SYNC pkt-active mode

Realtek Bluetooth :Get CONFG pkt-active mode

Realtek Bluetooth :3-wire config pattern resend : 2 , len: 10
Realtek Bluetooth :Get CONFG resp pkt-active mode

Realtek Bluetooth :H5 init finished

Realtek Bluetooth :config offset(f4),length(8)
Realtek Bluetooth :config baud rate to :4928002, hwflowcontrol:5f, 1
Realtek Bluetooth :config offset(27),length(1)
Realtek Bluetooth :config offset(fe),length(1)
Realtek Bluetooth :config offset(15b),length(4)
Realtek Bluetooth :config offset(1e3),length(1)
Realtek Bluetooth :Get config baud rate from config file:4928002
Realtek Bluetooth :Load FW OK
Realtek Bluetooth :RTK send HCI_VENDOR_READ_RTK_ROM_VERISION_Command

Realtek Bluetooth :Received reliable seqno 0 from card
Realtek Bluetooth :receive hci command complete event with command:1001

Realtek Bluetooth :Read RTK LMP version with Status:0
Realtek Bluetooth :gLmpVersion = 0x8723
Realtek Bluetooth :RTK send HCI_VENDOR_READ_RTK_ROM_VERISION_Command

Realtek Bluetooth :Received reliable seqno 1 from card
Realtek Bluetooth :receive hci command complete event with command:fc6d

Realtek Bluetooth :Read RTK rom version with Status:0
Realtek Bluetooth :rtk_hw_cfg.eversion = 1
Realtek Bluetooth :rtk_get_fw_project_id: opcode 0, len 1, data 1
Realtek Bluetooth :fw_ver 0x1e3ee40e, patch_num 2
Realtek Bluetooth :patch length is 0x5e90
Realtek Bluetooth :start offset is 0x4f00
Realtek Bluetooth :fw: exists, config file: exists
Realtek Bluetooth :baudrate in change speed command: 0x2 0x80 0x92 0x4 

Realtek Bluetooth :Received reliable seqno 2 from card
Realtek Bluetooth :receive hci command complete event with command:fc17

Realtek Bluetooth :Change BD Rate with status:0
Realtek Bluetooth :final_speed 1500000

Realtek Bluetooth :hw flow control enable
Realtek Bluetooth :iEndIndex:96  iLastPacketLen:71 iAdditionpkt:4

Realtek Bluetooth :hci_download_patch tx_index:0 rx_index: -1

Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:0
Realtek Bluetooth ERROR: patch timerout, retry:

```

Anyone seen this happen before?