lwfinger/rtl8723bs_bt #8: mixed repo

samica, 2016-02-19:
Hi,

Your repo seems to mix user space app with kernel space code.
Explanation:
hciattach.c
hciattach_rtk.c

are user space app.

But:
hci_h4.c
hci_ldisc.c
hci_rtk_h5.c
 should probably stay in <kernel_source>/drivers/bluetooth/
Then perhaps for hci_rtk_h5 you need to add a Kconfig entry as well as a Makefile entry.
Maybe a patch will be better...

Anyway, the problem is I am using the h5 (or the 3wire) and I still got "Realtek Bluetooth ERROR: H5 sync timed out".
I am thinking that maybe I need to patch the kernel with the rtk_h5 and not use h5 at all.
What do you say?

I'm gonna try it anyway and let you know the result.

lwfinger, 2016-02-19:
The repo has both kernel and user space stuff because a user needs both in order to have BT.

I did not write either of the parts and I have provided them only as a convenience.

Yes, some parts of this should be in the kernel, but I am never planning on having any part of that effort. All I can say is that if your device is not working with the existing code, then messing with the kernel is a waste of time. You should modify this code to find the problem. Then you might consider further actions. As my device does not have your problem, I cannot help.

samica, 2016-02-19:
Fair answer.

Since this is the only repo I could find for the sdio wifi/bt combo I am grateful you share the code with us.

I was mislead by your warning in the readme:

> WARNING: This driver needs the H5 protocol, which is built into kernels 3.12
and later. If you need to use an earlier kernel, raise an issue on this repo.

Btw, can you please state for what device you used this code with success.

samica, 2016-02-19:
Looking for H5 in the kernel I found CONFIG_BT_HCIUART_3WIRE that builds hci_h5.c.
But then there is rtk_h5.c in your repo...

Thus the misleading.

lwfinger, 2016-02-19:
If you look at the Makefile, that file is NEVER touched!! It it there for completeness and only needed for old kernels.

My device is the TW100 tablet.