lwfinger/rtl8723bs_bt #9: MeeGoPad T02 with Ubuntu 15.10, WiFi works, Bluetooth dont

Walrabe, 2016-02-23:
Hallo Larry,

i have a new MeeGoPad T02 and installed Ubuntu 15.10 on the internal eMMC. Then i downloaded/compiled/installed your rtl8723bs-master driver, all works fine (thanks for your work on it). After that, i downloaded/compiled/installed the _bt driver. Neither sudo ./rtk_hciattach -n -s 115200 /dev/ttyS4 rtk_h5 nor sudo ./rtk_hciattach -n -s 115200 /dev/ttyS5 rtk_h5 work for the MeeGoPad. I see ttyS4/ttyS5 in the dmesg and i modprobed rfkill-gpio manually, nothing works until now. Do you have any ideas? Thanks for your time!

Cheers, Chris Walrabe

lwfinger, 2016-02-23:
I have no clue. There is one report of similar problems. It is possible that the baud rate is not 115200, but something higher. There has been one report that once the firmware is loaded, the rate jumps to a much higher rate. See https://github.com/lwfinger/rtl8723bs_bt/issues/7, but there is no solution there.

Walrabe, 2016-02-24:
with this command: sudo ./rtk_hciattach -n -s 115200 /dev/ttyS4 rtk_h4 i get the following answer:

Realtek Bluetooth init uart with init speed:115200, final_speed:115200, type:HCI UART H4
Realtek Bluetooth :Realtek hciattach version 2.5 

Realtek Bluetooth :config offset(f4),length(8)
Realtek Bluetooth :config baud rate to :4928002, hwflowcontrol:5f, 1
Realtek Bluetooth :config offset(27),length(1)
Realtek Bluetooth :config offset(fe),length(1)
Realtek Bluetooth :config offset(15b),length(4)
Realtek Bluetooth :config offset(1e3),length(1)
Realtek Bluetooth :Get config baud rate from config file:4928002
Realtek Bluetooth :Load FW OK
Realtek Bluetooth ERROR: Check signature error!
Realtek Bluetooth :baudrate in change speed command: 0x2 0x80 0x92 0x4 

Realtek Bluetooth :H4 Change uart Baudrate after write 
Initialization timed out.

No device with: hcitool dev/scan

Cheers, Chris

jsno9, 2016-03-16:
hi Walrabe :
Have you solved this problem?
I meet the same problem as you.