lwfinger/rtl8723bs_bt #11: Can't initialize device: Invalid argument

zefie, 2016-05-02:
I am trying to get this to work with Android x86 on kernel 4.4.8. At first it was failing to send the 3wire packet, but I realized I had to prevent kernel modules btbcm, btintel, and btqca from loading. After that, it now reports this:

`Realtek Bluetooth init uart with init speed:115200, final_speed:115200, type:HCI UART H5
Realtek Bluetooth :Realtek hciattach version 2.5 

Realtek Bluetooth :3-wire sync pattern resend : 1, len: 8

Realtek Bluetooth :Get SYNC Resp Pkt

Realtek Bluetooth :Get SYNC pkt-active mode

Realtek Bluetooth :3-wire config pattern resend : 1 , len: 10
Realtek Bluetooth :Get CONFG pkt-active mode

Realtek Bluetooth :Get CONFG resp pkt-active mode

Realtek Bluetooth :H5 init finished

Realtek Bluetooth :config offset(f4),length(8)
Realtek Bluetooth :config baud rate to :4928002, hwflowcontrol:5f, 1
Realtek Bluetooth :config offset(27),length(1)
Realtek Bluetooth :config offset(fe),length(1)
Realtek Bluetooth :config offset(15b),length(4)
Realtek Bluetooth :config offset(1e3),length(1)
Realtek Bluetooth :Get config baud rate from config file:4928002
Realtek Bluetooth :Load FW OK
Realtek Bluetooth :RTK send HCI_VENDOR_READ_RTK_ROM_VERISION_Command

Realtek Bluetooth :Received reliable seqno 0 from card
Realtek Bluetooth :receive hci command complete event with command:1001

Realtek Bluetooth :Read RTK LMP version with Status:0
Realtek Bluetooth :gLmpVersion = 0x8723
Realtek Bluetooth :RTK send HCI_VENDOR_READ_RTK_ROM_VERISION_Command

Realtek Bluetooth :Received reliable seqno 1 from card
Realtek Bluetooth :receive hci command complete event with command:fc6d

Realtek Bluetooth :Read RTK rom version with Status:0
Realtek Bluetooth :rtk_hw_cfg.eversion = 1
Realtek Bluetooth :rtk_get_fw_project_id: opcode 0, len 1, data 1
Realtek Bluetooth :fw_ver 0x1e3ee40e, patch_num 2
Realtek Bluetooth :patch length is 0x5e90
Realtek Bluetooth :start offset is 0x4f00
Realtek Bluetooth :fw: exists, config file: exists
Realtek Bluetooth :baudrate in change speed command: 0x2 0x80 0x92 0x4 

Realtek Bluetooth :Received reliable seqno 2 from card
Realtek Bluetooth :receive hci command complete event with command:fc17

Realtek Bluetooth :Change BD Rate with status:0
Realtek Bluetooth :final_speed 1500000

Realtek Bluetooth :hw flow control enable
Realtek Bluetooth :iEndIndex:96  iLastPacketLen:71 iAdditionpkt:4

Realtek Bluetooth :hci_download_patch tx_index:0 rx_index: -1

Realtek Bluetooth :Received reliable seqno 3 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 0

Realtek Bluetooth :hci_download_patch tx_index:1 rx_index: 0

Realtek Bluetooth :Received reliable seqno 4 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 1

Realtek Bluetooth :hci_download_patch tx_index:2 rx_index: 1

Realtek Bluetooth :Received reliable seqno 5 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 2
[... 472 lines cut ...]

Realtek Bluetooth :Received reliable seqno 4 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 97

Realtek Bluetooth :hci_download_patch tx_index:98 rx_index: 97

Realtek Bluetooth :Received reliable seqno 5 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 98

Realtek Bluetooth :hci_download_patch tx_index:99 rx_index: 98

Realtek Bluetooth :Received reliable seqno 6 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 99

Realtek Bluetooth :Send FW last command
Realtek Bluetooth :hci_download_patch tx_index:100 rx_index: 99

Realtek Bluetooth :Received reliable seqno 7 from card
Realtek Bluetooth :rtk_hw_cfg.rx_index 100

Realtek Bluetooth :Init Process finished
**Can't set line discipline: Invalid argument
Can't initialize device: Invalid argument**`

Any ideas what would cause this?

lwfinger, 2016-05-02:
No, I have no ideas. All I know is that the code works on my TW100. I have no idea what the differences might be on any other tablet, other than to note that the hardware seems to differ a lot from device to device.

youling257, 2016-05-03:
see here，https://groups.google.com/forum/#!topic/Android-x86/kPBpj3CoXW4 ，

zefie, 2016-05-05:
I think the key here is the version of the rtl8723bs wifi driver. I am now using the latest from https://github.com/hadess/rtl8723bs and it is back to timing out sending 3wire command. Knowing which commit and/or version of the driver this was tested against would be of help.