lwfinger/rtl8723bs_bt #12: Works on Lenovo miix 3-830 with another firmware

Ilimos, 2016-05-24:
With the git bluetooth driver hci0 was set up properly but scan/discovery by other devices was not giving any results. I tried following 3 cases 
1- git fw+config:  hci0 ok but no scan
2- fw+config from windows installation : error parsing config - no interface
3- fw from windows + config from git: hci0 ok - scan ok

fw from windows is slightly larger at 51364. Can send over if anybody is interested.
Tested on baytrail tablet Lenovo miix 3-830 on Arch with vanilla 4.1.23 kernel patched for rtl8723bs wifi module. For BT to work I also added the acpi id (OBDA8723) of the BT module to rfkill-gpio.c to activate the module.

Many thanks for the driver.

lwfinger, 2016-05-26:
If you send the firmware file to Larry.Finger@lwfinger.net, I will add it to the git repo. The new one seems to be an improvement. If you also send the kernel patch, I will make that available. BTW, that patch should also be sent to Johannes Berg <johannes@sipsolutions.net> for inclusion in the newest kernels. If you need help submitting a kernel patch, let me know.

kahrloz, 2016-07-10:
@marc-wx  how did you get the fw file?

marc-wx, 2016-07-14:
I copied it from the Windows installation of the tablet :
/Windows/System32/rtl8723b_mp_bt40_fw_asic_rom_patch.bin
File is approx 50 kB. You need to rename it to rtlbt_fw so that it works with the driver from lwfinger.

lwfinger, 2016-07-16:
I do not have a Windows installation on my tablet. Please send a copy of the new firmware to Larry.Finger@lwfinger.net. Thanks.

marc-wx, 2016-07-17:
Done!

lwfinger, 2016-07-17:
Thanks. I added the file to the repo as rtlbt_fw_new. That way it will be available for any that need it.

closed 2016-07-17 by lwfinger