lwfinger/rtl8723bs_bt #13: Bluetooth not working on PCG02U

MatejKovacic, 2016-06-23:
Hi,

there is a cheap Ubuntu Stick PC called PCG02U, which uses RTL8723BS chip. However, bluetooth is not working. The problem is described here:
http://www.cnx-software.com/2016/05/26/updating-star-cloud-pcg02u-to-ubuntu-16-04-with-wifi-and-hdmi-audio-support/

Bluetooth is being located on /dev/ttyS4, but connection timeouts. Changing the baudrate from 115200 in the script to 2764800 does not help.

Any idea?

MatejKovacic, 2016-06-24:
Okay, now I tried to run the drivers by myself. I have Ubuntu Stick PC PCG02U, with (original) Ubuntu 14.04.4 LTS. Kernel is 3.16.0-71-generic.

I install the drivers with the commands:
```
git clone https://github.com/lwfinger/rtl8723bs_bt
cd rtl8723bs_bt/
make
sudo make install
sudo ~/rtl8723bs_bt/start_bt.sh
```

Now it says:

```
No BT TTY device has been found
Either this computer has no BT device that uses hciattach, or
Your kernel does not have module 8250_dw configured.
Note: The configuration variable is CONFIG_SERIAL_8250_DW.
```
I tried to load a module 8250_dw, but it is the same. Any idea?

lwfinger, 2016-06-24:
First of all, does module 8250_dw loading? You can use the lsmod command to see if that is true.

Secondly, do you see any mention of the extra tty's in dmesg. The command 'dmesg | grep tty' should give that answer.

MatejKovacic, 2016-06-24:
Hi,

sorry, it seems there has been false alarm. **After reboot**, it seems to be working (in fact, it is not, as described below):
```
sudo ~/rtl8723bs_bt/start_bt.sh
Using device /dev/ttyS4 for Bluetooth
```
`dmesg | grep tty` says:
```
[    0.000000] console [tty0] enabled
[    1.160972] 00:02: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
[    5.355440] 80860F0A:00: ttyS4 at MMIO 0x90941000 (irq = 39, base_baud = 2764800) is a 16550A
[    5.356217] 80860F0A:01: ttyS5 at MMIO 0x90947000 (irq = 40, base_baud = 2764800) is a 16550A
```

However, Ubuntu does not show me a Bluetooth device. Also, `hcitool dev` does not list any local devices. Any idea what could be wrong?

MatejKovacic, 2016-06-27:
BTW, 8250_dw module is loaded:
```
lsmod | grep 8250_dw
8250_dw                13551  0
```

When running this command:
`sudo rtk_hciattach -n -s 115200 /dev/ttyS4 rtk_h5`
I got this output:
```
Realtek Bluetooth init uart with init speed:115200, final_speed:115200, type:HCI UART H5
Realtek Bluetooth :Realtek hciattach version 3.1 

Realtek Bluetooth :3-wire sync pattern resend : 1, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 2, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 3, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 4, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 5, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 6, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 7, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 8, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 9, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 10, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 11, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 12, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 13, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 14, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 15, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 16, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 17, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 18, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 19, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 20, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 21, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 22, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 23, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 24, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 25, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 26, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 27, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 28, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 29, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 30, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 31, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 32, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 33, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 34, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 35, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 36, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 37, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 38, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 39, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 40, len: 8

Realtek Bluetooth ERROR: H5 sync timed out

```

MatejKovacic, 2016-06-27:
Another info (strace): http://pastebin.com/QubcYMvu