lwfinger/rtl8723bs_bt #15: Not working on Lenovo miix 3 830

Haxk20, 2016-07-18:
I have rtl8723bs wifi driver from hadess and tried this since i installed ubuntu 16.04 LTS on my device but BT dont show up in ubuntu fo i need your wifi driver ? Or what i shoild try ?

Haxk20, 2016-07-18:
And wifi is working with hadess driver totally normal no problems only with BT

lwfinger, 2016-07-18:
Nothing works without a driver.

Haxk20, 2016-07-18:
I have your driver installed

Haxk20, 2016-07-18:
BT driver

Haxk20, 2016-07-18:
Can you help me ?

lwfinger, 2016-07-18:
As I know I do not have your tablet, I'm not sure what I can do.

You should review all the issues in this repo for suggestions. What happens when you run start_bt.sh?

What ttys are created? The command "dmesg | grep tty" will help answer that.

Haxk20, 2016-07-19:
dmesg | grep tty : 
[    0.000000] console [tty0] enabled
[    1.656107] 00:02: ttyS0 at I/O 0x3f8 (irq = 199, base_baud = 115200) is a 16550A
[    4.823170] systemd[1]: Created slice system-getty.slice.
[    5.410271] 80860F0A:00: ttyS4 at MMIO 0x90947000 (irq = 191, base_baud = 2764800) is a 16550A
[    5.410512] 80860F0A:01: ttyS5 at MMIO 0x9094d000 (irq = 192, base_baud = 2764800) is a 16550A
and start_bt.sh shows using device ttyS4 as bluetooth
and lsmod dont find   rtl8723_bs only r8723bs which is wifi

lwfinger, 2016-07-19:
There is NO module for BT. Everything is run in user space. Please follow the instructions in "readme.txt".

Haxk20, 2016-07-19:
I have done installation and that thing when i start bt from start_bt.sh its not working i will try to recompile kernel with the fixed rfkill_gpio.c maybe that will work

Haxk20, 2016-07-19:
and output of hciattach.txt : /home/haxk20/rtl8723bs_bt/start_bt.sh: line 35: ./rtk_hciattach: No such file or directory 
i dont know but it seems like its canoot find the rtk_hciattach

kahrloz, 2016-07-19:
In terminal go to the rtl8723bs_bt and do a "make clean" command

kahrloz, 2016-07-19:
And start over with the "make", and "make install ", once finished with the file manager go to the rtl8723bs_bt directory and right click rtk_hciattach and make it executable under properties

lwfinger, 2016-07-19:
It will be executable. That step is not required; however, you MUST be in the directory with rtk_hciattach, otherwise you will need to edit start_bt.sh.

Haxk20, 2016-07-20:
will do toomorrow since today i dont have much time

Haxk20, 2016-07-25:
Ok so trued that not working still says no such file or directory and yes im in folder where rtk_hciattach is but still problem

Haxk20, 2016-07-25:
And still cannot recompile kernel because its need lots of space to compile should somebody here compile the kernel with edited rfkill_gpio.c to support your moduke i saw that in other issue

Haxk20, 2016-12-31:
working closing isue

closed 2016-12-31 by Haxk20

vomitodegato, 2018-06-23:
hello,
what's wrong for me please?

`dmesg | grep tty
[    0.000000] console [tty0] enabled
[    2.478155] 00:01: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
[    5.740667] 8086228A:00: ttyS4 at MMIO 0x91821000 (irq = 39, base_baud = 2764800) is a 16550A
[    5.741287] serial serial0: tty port ttyS4 registered
[    5.748585] 8086228A:01: ttyS5 at MMIO 0x9181f000 (irq = 40, base_baud = 2764800) is a 16550A
`

i've got the BT problem only (wifi is ok).

Shneider SCL141CTP    (Lubuntu, sound: bytcr-rt5651, mother: Atom-celeron-pentium X5-E8000 Series PCI , wifi-blutooth :RTL8723BS)