lwfinger/rtl8723bs_bt #19: Autostart

Haxk20, 2016-12-31:
How to make the script to autorun at boot ?

madivak, 2017-02-22:
implement it on the start-stop-daemon That's how i did. Though i havent found a way to send and receive files.

Haxk20, 2017-04-08:
Its been a long time i got fixed a lot of stuff on my tablet and thats this autostart problem  if anybody found a way post a tutorial here please

madivak, 2017-04-09:
Well I managed to sort all of my Bluetooth problems including connecting and transferring of files. But am using build root and a different hardware. I.e an allwinner A20 board. Will post a tutorial on monday

madivak, 2018-01-28:
I made a few changes. Added my own readme. Check below. That's what I did.

https://github.com/madivak/rtl8723bs_bt?files=1

madivak, 2018-01-29:
i first initialize the bluetooth wake and wake-host. using this script.

`#!/bin/bash
#configuration to get pin BT_WAKE pin to register, be gpio_out and turned on

echo "239"  > /sys/class/gpio/export   
echo "in"  > /sys/class/gpio/gpio239/direction  
echo "rising"  > /sys/class/gpio/gpio239/edge

echo "238" > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio238/direction
echo "1" > /sys/class/gpio/gpio238/value`

Then inititialize the bluetooth interface with the following NOTE: My bluetooth is attached to serial /dev/ttyS2

`#!/bin/bash
#launch Bluetooth driver

rtk_hciattach -n -b -s 115200 /dev/ttyS2 rtk_h5 > hciattach.txt 2>&1 &

/bin/sleep 5		#allow hci0 interface to initialise

hciconfig               #show if hci inteface is up

hciconfig hci0 up       #bring up the hci inteface

hciconfig -a            #show if hci inteface is up

hciconfig hci0 piscan   #Make my device discoverable

hcitool scan            #scan available BT signals`

closed 2021-03-16 by Haxk20