lwfinger/rtl8723bs_bt #21: HOW DO I CROSS COMPILE THIS DRIVER

madivak, 2017-02-07:
i have cross compiled the rtl8723bs WiFi driver from https://github.com/hadess/rtl8723bs/issues and got the WiFi working.
But i dont know how approach this bt driver since its not a module. Can someone please point this out.
i have done 
`
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf-`

which generates rtk_hciattach and hciattach_rtk.o
but the bit for make install?

closed 2017-02-07 by madivak

madivak, 2017-02-17:
ended up cross compiling using the edited [https://github.com/NextThingCo/rtl8723bs_bt] but havent yet go the BT working.

MalakiLab, 2017-04-06:
It's only a few lines of codes, why would you cross compile it? And it is a kernel module, but compiled outside, then when you do "make install" it's installing it where it's supposed to be in your /lib/modules/kernel-XYZ/place-it-should-be/. So well, in any case, you should compile it on the target device and not cross compile on another machine in the first place, since you'll need make to install it anyway.

lwfinger, 2017-04-06:
You need to cross compile the code if and only if the kernel is cross compiled. Any other combination, and the modules will not load.

madivak, 2017-04-06:
I managed to get the Bluetooth working fine. Even tested with transfer of a few files. Edited the make file a little bit then cross compiled it and it worked on a sun7i arm device. Thanks for the help though.