lwfinger/rtl8723bs_bt #26: Not working on Ubuntu 17.10

futbarista, 2017-12-04:
I've done everything with was written in readme.txt. No errors at writing commands, everything's fine but bluetooth still doesn't work.

zhongtianxie, 2017-12-28:
yes, my ubuntu 17.10 also can't work

djl1965, 2018-03-08:
Also doesn't work for me with Ubuntu 17.10 (using AzulleTech Quantum Access LAN)
Tried both versions of your supplied firmware.
Get the common error...
Realtek Bluetooth ERROR: H5 sync timed out.
Any clues what to try next?

zhongtianxie, 2018-03-09:
HI I'd met this issue on android-x86 with kernel 4.14, The issue root cause
due to new kernel version(kernel-4.11 older) add a subsystem named Serdev;
you can read this article
https://www.phoronix.com/scan.php?page=news_item&px=Linux-4.11-Serial-Bus-SERDEV

This driver for rtl8723bs from https://github.com/lwfinger/rtl8723bs_bt is
not serdev device;
the patch details you can find on
https://github.com/jwrdegoede/linux-sunxi/commit/222cd00fc3a8ca016cb1d253bb8866b706505c35
;

noticed the commit msg :

This causes the /dev/tty# node for uarts referenced by a UartSerialBusV2
to disappear, which breaks support for bluetooth over uart which needs
a*btattach from userspace* and does not yet have an in kernel hci
serdev
driver.

One example of a bluetooth over uart device which lacks such a driver and
thus no longer works is the RTL8723BS's bluetooth part, this commit adds
an exception for this device to acpi_is_serial_bus_slave(), so that it
gets treated as before, allowing it to work with the userspace solution
for now.

about  *btattach from userspace *tools, you can find some help from
https://osdn.net/projects/android-x86/scm/git/external-bluetooth-bluez/commits/de7b62e01ae65b3c7a8412515996387fbc431834

good luck.

frakswe, 2018-09-21:
thanks zhongtianxie for explaining why they (dev/ttyS4) vanish after boot, shame it's still not fixed in 4.18.8 mainline. i'm not good enough to use your patch but thanks anyway for helping.

svenha, 2019-02-13:
Just a quick update: Under Ubuntu 19.04 (installed from a daily ISO), bluetooth works out-of-the-box. Use `bluetoothctl` (or other tools) to pair and setup your device. The required firmware files are located in /usr/lib/firmware/rtl_bt/.