lwfinger/rtl8723bs_bt #27: Bluetooth not powered up on 4.15.2 kernel

marc-wx, 2018-02-18:
Hello
Up to kernel 4.14.13 I used the patch below to power up the bt chip at startup before launching the hciattach script : worked fine. After compiling 4.15.2 the hack is not working anymore and 'rfkill list' does not list the bt device. Anybody noticed the same issue and is there another way to fire up the device?
Cheers

diff --git a/net/rfkill/rfkill-gpio.c b/net/rfkill/rfkill-gpio.c
index 76c01cb..6e7e0ca 100644
--- a/net/rfkill/rfkill-gpio.c
+++ b/net/rfkill/rfkill-gpio.c
@@ -161,6 +161,7 @@ static int rfkill_gpio_remove(struct platform_device *pdev)
 
 #ifdef CONFIG_ACPI
 static const struct acpi_device_id rfkill_acpi_match[] = {
+	{ "OBDA8723",RFKILL_TYPE_BLUETOOTH },
 	{ "BCM4752", RFKILL_TYPE_GPS },
 	{ "LNV4752", RFKILL_TYPE_GPS },
 	{ },

lwfinger, 2018-02-18:
You should post this question on the Bluetooth mailing list - linux-bluetooth@vger.kernel.org. On that list it will be seen by people who know a lot about Bluetooth.

closed 2018-02-18 by lwfinger

Haxk20, 2018-03-09:
This is due to serdev. Please check issue named doesn't work on Ubuntu 17.10 it is explained in there + patch for it

marc-wx, 2018-03-11:
Thanks! I tried to google for Ubuntu 17.10 + bluetooth + serdev + rtl8723 etc... No result, do you have any link for the patch?

Haxk20, 2018-03-13:
https://github.com/jwrdegoede/linux-sunxi/commit/222cd00fc3a8ca016cb1d253bb8866b706505c35
Here.

sheldonmaschmeyer, 2018-07-23:
Please excuse my ignorance but on, for instance Manjaro-arch, where is "/drivers/acpi/scan.c" ?