lwfinger/rtl8723bs_bt #28: Manjaro-Arch Linux Kernel 4.17

sheldonmaschmeyer, 2018-07-07:
Hi, I am trying to get bluetooth to work on Manjaro-Arch Linux Kernel 4.17. Have a CHUWI Hi10 Pro (CWI529, HQ64G4216xxxx). I followed the instructions with no errors. Start_bt.sh when run prints "Using device /dev/ttyS1 for Bluetooth" But when I try to start Bluetooth Manager, states "no Bluetooth Adapters Detected" and "sudo hwdetect --show-bluetooth" shows nothing.
Any advise appreciated.

lwfinger, 2018-07-07:
That script can not always find the correct serial device. In addition, it has to guess at the starting rate.

Have you looked at file hciattach.txt?

I thought that kernels as new as 4.17 had code built in for serial BT devices.

sheldonmaschmeyer, 2018-07-07:
No bluetooth adapters are being detected by default. 4.17 might have code for serial BT but, not the driver for my tablet.
$ dmesg -t | grep tty | grep MMIO | cut -b 14-18
ttyS1
ttyS2

For ttyS1 - hciattach.txt: 
"Can't open serial port: No such file or directory
Can't initialize device: Invalid argument"

For ttyS2 - hciattach.txt:
Realtek Bluetooth init uart with init speed:115200, final_speed:115200, type:HCI UART H5
Realtek Bluetooth :Realtek hciattach version 2.5 

Realtek Bluetooth :3-wire sync pattern resend : 1, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 2, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 3, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 4, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 5, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 6, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 7, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 8, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 9, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 10, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 11, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 12, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 13, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 14, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 15, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 16, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 17, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 18, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 19, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 20, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 21, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 22, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 23, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 24, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 25, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 26, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 27, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 28, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 29, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 30, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 31, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 32, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 33, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 34, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 35, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 36, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 37, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 38, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 39, len: 8

Realtek Bluetooth :3-wire sync pattern resend : 40, len: 8

Realtek Bluetooth ERROR: H5 sync timed out

For ttyS2 I simplified your script:
"
#!/bin/bash
#
# Shell script to install Bluetooth firmware and attach BT part of
# RTL8723BS
#

TTY="/dev/ttyS2"
echo "Using device $TTY for Bluetooth"

if [ ! -f /lib/firmware/rtl_bt/rtlbt_config ];
then
	mkdir -p /lib/firmware/rtl_bt/
	cp rtlbt_* /lib/firmware/rtl_bt/.
fi
./rtk_hciattach -n -s 115200 $TTY rtk_h5 > hciattach.txt 2>&1 & 
"

I also tried:
sudo cp rtlbt_fw_new /lib/firmware/rtl_bt/rtlbt_fw

$ sudo hwdetect --show-bluetooth --> displays nothing
$ sudo systemctl stop bluetooth
$ sudo systemctl enable bluetooth
$ sudo systemctl start bluetooth

Thank you for helping :-)

lwfinger, 2018-07-08:
Please run the following to determine what ttyS devices are on your system:
ls /dev/ttyS*

In addition, please copy the output of the dmesg command to some pastebin location and post the link here.

sheldonmaschmeyer, 2018-07-09:
$ /dev/ttyS*
/dev/ttyS0  /dev/ttyS2  /dev/ttyS3
$ dmesg
[    0.000000] Linux version 4.17.4-1-MANJARO (builduser@development) (gcc version 8.1.1 20180531 (GCC)) #1 SMP PREEMPT Tue Jul 3 18:11:09 UTC 2018
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.17-x86_64 root=UUID=eb1627ff-dc80-4044-bf9b-6e7b5f34c1cf rw quiet resume=UUID=7b02da92-ae35-4ed8-8aad-f070fdd20161 fbcon=rotate:1
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000008efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000008f000-0x000000000008ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000090000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001effffff] usable
[    0.000000] BIOS-e820: [mem 0x000000001f000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x000000007ad01fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ad02000-0x000000007ad31fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007ad32000-0x000000007ad5ffff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007ad60000-0x000000007b5b2fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007b5b3000-0x000000007b870fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b871000-0x000000007b8f9fff] type 20
[    0.000000] BIOS-e820: [mem 0x000000007b8fa000-0x000000007bffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000e3ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000feafffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed03000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed06000-0x00000000fed06fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed09fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1cfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fedbffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffa00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000017fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by American Megatrends
[    0.000000] efi:  ESRT=0x7ad30000  ACPI=0x7ad3c000  ACPI 2.0=0x7ad3c000  SMBIOS=0x7b770000  SMBIOS 3.0=0x7b76f000 
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: CHUWI INNOVATION AND TECHNOLOGY(SHENZHEN)CO.LTD Hi10 pro tablet/Cherry Trail CR, BIOS 5.11 11/10/2016
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x180000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000 mask F80000000 write-back
[    0.000000]   1 base 07E000000 mask FFE000000 uncachable
[    0.000000]   2 base 07D000000 mask FFF000000 uncachable
[    0.000000]   3 base 07C800000 mask FFF800000 uncachable
[    0.000000]   4 base 100000000 mask F80000000 write-back
[    0.000000]   5 base 07C400000 mask FFFC00000 uncachable
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000000] total RAM covered: 4036M
[    0.000000] Found optimal setting for mtrr clean up
[    0.000000]  gran_size: 64K 	chunk_size: 64M 	num_reg: 6  	lose cover RAM: 0G
[    0.000000] e820: update [mem 0x7c400000-0xffffffff] usable ==> reserved
[    0.000000] e820: last_pfn = 0x7c000 max_arch_pfn = 0x400000000
[    0.000000] esrt: ESRT header is not in the memory map.
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [        (ptrval)] 98000 size 24576
[    0.000000] BRK [0x11366d000, 0x11366dfff] PGTABLE
[    0.000000] BRK [0x11366e000, 0x11366efff] PGTABLE
[    0.000000] BRK [0x11366f000, 0x11366ffff] PGTABLE
[    0.000000] BRK [0x113670000, 0x113670fff] PGTABLE
[... 767 lines cut ...]
[   35.830362] usb 1-3: New USB device found, idVendor=05e3, idProduct=0608, bcdDevice=85.36
[   35.830383] usb 1-3: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[   35.830395] usb 1-3: Product: USB2.0 Hub
[   35.832950] hub 1-3:1.0: USB hub found
[   35.833277] hub 1-3:1.0: 4 ports detected
[   36.120399] usb 1-3.1: new low-speed USB device number 6 using xhci_hcd
[   36.218684] usb 1-3.1: New USB device found, idVendor=1c4f, idProduct=0063, bcdDevice= 1.10
[   36.218705] usb 1-3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   36.218716] usb 1-3.1: Product: USB Keyboard
[   36.218726] usb 1-3.1: Manufacturer: LIZHICHIP
[   36.231798] input: LIZHICHIP USB Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.0/0003:1C4F:0063.0005/input/input16
[... cut ...]
[   36.598998] hid-generic 0003:3938:1047.0007: input,hidraw2: USB HID v1.10 Keyboard [MOSART Semi. 2.4G Wireless Mouse] on usb-0000:00:14.0-3.2/input0
[   36.604560] input: MOSART Semi. 2.4G Wireless Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.1/0003:3938:1047.0008/input/input19
[   36.661938] hid-generic 0003:3938:1047.0008: input,hiddev1,hidraw3: USB HID v1.10 Mouse [MOSART Semi. 2.4G Wireless Mouse] on usb-0000:00:14.0-3.2/input1
[  508.482503] RTL8723BS: send eapol packet
[  508.488120] RTL8723BS: set group key camid:6, addr:90:50:ca:17:5d:58, kid:1, type:AES

sheldonmaschmeyer, 2018-07-10:
Hi, since hciattach is depreciated, I tried to use btattach (the replacement):
$ sudo btattach -N -S 115200 --bredr /dev/ttyS2 -P 3wire
Attaching Primary controller to /dev/ttyS2
Switched line discipline from 0 to 15
Failed to get device id: Protocol driver not attached
No controller attached

svenha, 2018-09-10:
Does a kernel bug report exist for this issue?

Enginecrafter77, 2018-09-10:
I doubt so. I analyzed the commits on the r8723bs driver in staging tree, and it seems there is little to no work being done on it, sooo... So you are probably not going to see it anytime soon, unless you are C guru and willing to move throught the commit submitting hassle.

lwfinger, 2018-09-10:
Did you also look at the commits for drivers/bluetooth/btrtl.c? That is where any BT changes would be made. I know there are significant additions of code for serial devices in kernel 4.19-rc1, thus you probably should get the source for the mainline kernel and test with it. If your device does not work with that kernel (no external drivers needed), then that should be reported to the mailing list linux-bluetooth@vger.kernel.org.

FYI, my only part of this activity is to take drivers sent to me by Realtek and verify that they build. I do no testing of such drivers, thus I have no means of fixing them, other than build errors. Others are involved in the current work to ensure that the devices work with btrtl.ko in conjunction with btusb.ko.

Enginecrafter77, 2018-09-10:
I looked it up today in the kernel tree, and in 4.18 appears to be nothing related to the rtl8723BS in the drivers/bluetooth/btrtl.c, but in the upcoming release, 4.19 (rc3 currently), the btrtl.c appars to be supplied with some rtl8723BS-relevant code, so we could see some improvements soon I think. I currently don't have enought machines to test it on, so I can't test it anytime soon.

svenha, 2018-09-11:
I switched to kernel
4.19.0-1-MANJARO #1 SMP PREEMPT Thu Sep 6 18:10:15 UTC 2018 x86_64 GNU/Linux
without any improvement. i.e. the ttyS1 which is selected by the script start_bt.sh is not existing before I call it.

Some dumb questions:

Should I recreate this ttyS1 or will it be deleted by the (incorrect) boot process? It the former, how?

Do I have to uninstall some files from this repo?

All firmware files in /lib/firmware/rtl_bt have a .bin suffix except for the ones from this repo. Should they be renamed? I tried and followed the scheme in the Orochimarufan repo - no change.

Any findings whether rtlbt_fw  or rtlbt_fw_new should be preferred?

Enginecrafter77, 2018-09-11:
No, manually creating the device file probably won't help you. At least it didn't for me. And... Let the files be as they are. Renaming them only confuses the software, so don't do it.

svenha, 2018-09-21:
I can confirm that a missing device file (ttyS1 in my case) is a fatal symptom after booting. It is explained in issue #26.

emelcher, 2018-10-24:
In kernel 4.19, no patch is needed. Neither start_bt.sh nor rtk_hciattach are needed.
**Kernel .config:**
CONFIG_SERIAL_8250_DW=m
CONFIG_RFKILL_GPIO=m
CONFIG_SERIAL_DEV_BUS=y
CONFIG_BT_HCIUART_RTL=y
**Firmware:**
copy rtlbt_fw to /usr/lib/firmware/rtl_bt/rtl8723bs_fw.bin
copy rtlbt_config to /usr/lib/firmware/rtl_bt/rtl8723bs_config.bin
cd /usr/lib/firmware/rtl_bt; ln -s rtl8723bs_config.bin rtl8723bs_config-OBDA8723.bin
**Test** (as root):
hciconfig -a
hciconfig hci0 up
hciconfig hci0 piscan
(as user):
hcitool scan

svenha, 2018-10-26:
The kernel .config for manjaro seems to be ok (checked for kernel 4.19), except that CONFIG_BT_HCIUART_RTL is not set. Is this a problem?

emelcher, 2018-10-26:
Yes, it is a problem. Even CONFIG_BT_HCIUART_RTL=m does not work, CONFIG_BT_HCIUART_RTL=y is needed. In my distribution (CentOS7) the commands hciconfig and hcitool are in a package named bluez.

lwfinger, 2018-10-26:
If your kernel has CONFIG_BT_HCIUART_RTL=m, you might have to manually load module hci_uart. The serial BT device would not be able to be found by a bus scan. I cannot think of any other reason for the "m" option to fail.

svenha, 2018-11-07:
Manjaro kernel 4.19.1 contains all required modules. Instead of hciconfig, one can use bluetoothctl, see https://wiki.archlinux.org/index.php/Bluetooth_keyboard .

sheldonmaschmeyer, 2019-01-19:
I upgraded Ubuntu 18.10 Kernel to 4.20 
I then followed the instructions below but ran into an issue:
git https://github.com/lwfinger/rtl8723bs_bt.git
cd rtl8723bs_bt
cp rtlbt_fw /usr/lib/firmware/rtl_bt/rtl8723bs_fw.bin
cp rtlbt_config /usr/lib/firmware/rtl_bt/rtl8723bs_config.bin
cd /usr/lib/firmware/rtl_bt
ln -s rtl8723bs_config.bin rtl8723bs_config-OBDA8723.bin

Reboot.

As a normal user configure bluetooth (only needed once) by starting bluetoothctl and then entering the following command at its prompt:
list
power on
agent KeyboardOnly (sometimes already done)
default-agent
pairable on
(turn your bluetooth device on and put into pairing mode)
scan on
pair 01:02:03:04:05:06 (use the id of your device here and below)
paired-devices (just for information)
trust 01:02:03:04:05:06
connect 01:02:03:04:05:06

Your bluetooth device should work now.

If you need to reconnect (after a reboot etc.), you only need power on and connect.
You can put this in a shell script like this:
echo "power on" | bluetoothctl
echo "connect 01:02:03:04:05:06" | bluetoothctl --timeout 10

The issue is at the connect stage, i't successfully pairs however, devices (laptop and android phone) don't connect to the tablet.
However, Unified Remotes using SDP bluetooth protocol does work (does not require device pairing) using the below instructions:
In  /etc/systemd/system/dbus-org.bluez.service:
ExecStart=/usr/lib/bluetooth/bluetoothd --compat
ExecStartPost=/bin/chmod 777 /var/run/sdp
So, progress has been made however, still cannot use with bluetooth headphones, android phones (except for Unified Remotes) and, between two computers.

svenha, 2019-02-13:
If you want to pair non-keyboard devices, the command `agent KeyboardOnly` does not look right.