lwfinger/rtl8723bs_bt #31: Realtek Bluetooth ERROR: patch timerout, retry

xuxianyin5490, 2018-08-25:
Hi，
    I am new for rtl8723bs, what's wrong for me？ Can you help me? @lwfinger 

root@Telescope:~# ./rtk_hciattach -n -s 115200 /dev/ttyO2  rtk_h5
Realtek Bluetooth init uart with init speed:115200, final_speed:115200, type:HCI UART H5
Realtek Bluetooth :Realtek hciattach version 2.5 

Realtek Bluetooth :3-wire sync pattern resend : 1, len: 8

Realtek Bluetooth :Get SYNC Resp Pkt

Realtek Bluetooth :3-wire config pattern resend : 1 , len: 10
Realtek Bluetooth :Get SYNC pkt-active mode

Realtek Bluetooth :Get SYNC pkt-active mode

Realtek Bluetooth :Get CONFG pkt-active mode

Realtek Bluetooth :3-wire config pattern resend : 2 , len: 10
Realtek Bluetooth :Get CONFG resp pkt-active mode

Realtek Bluetooth :H5 init finished

Realtek Bluetooth :config offset(f4),length(8)
Realtek Bluetooth :config baud rate to :14c05202, hwflowcontrol:5f, 1
Realtek Bluetooth :config offset(27),length(1)
Realtek Bluetooth :config offset(fe),length(1)
Realtek Bluetooth :config offset(15b),length(4)
Realtek Bluetooth :config offset(1e3),length(1)
Realtek Bluetooth :Get config baud rate from config file:14c05202
Realtek Bluetooth :Load FW OK
Realtek Bluetooth :RTK send HCI_VENDOR_READ_RTK_ROM_VERISION_Command

Realtek Bluetooth :Received reliable seqno 0 from card
Realtek Bluetooth :receive hci command complete event with command:1001

Realtek Bluetooth :Read RTK LMP version with Status:0
Realtek Bluetooth :gLmpVersion = 0x8723
Realtek Bluetooth :RTK send HCI_VENDOR_READ_RTK_ROM_VERISION_Command

Realtek Bluetooth :Received reliable seqno 1 from card
Realtek Bluetooth :receive hci command complete event with command:fc6d

Realtek Bluetooth :Read RTK rom version with Status:0
Realtek Bluetooth :rtk_hw_cfg.eversion = 1
Realtek Bluetooth :rtk_get_fw_project_id: opcode 0, len 1, data 1
Realtek Bluetooth :fw_ver 0x1e3ee40e, patch_num 2
Realtek Bluetooth :patch length is 0x5e90
Realtek Bluetooth :start offset is 0x4f00
Realtek Bluetooth :fw: exists, config file: exists
Realtek Bluetooth :baudrate in change speed command: 0x2 0x52 0xc0 0x14 

Realtek Bluetooth :Received reliable seqno 2 from card
Realtek Bluetooth :receive hci command complete event with command:fc17

Realtek Bluetooth :Change BD Rate with status:0
Realtek Bluetooth :final_speed 115200

Realtek Bluetooth :hw flow control enable
Realtek Bluetooth :iEndIndex:96  iLastPacketLen:71 iAdditionpkt:4

Realtek Bluetooth :hci_download_patch tx_index:0 rx_index: -1

Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:0
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:1
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:2
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:3
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:4
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:5
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:6
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:7
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:8
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:9
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:10
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:11
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:12
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:13
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:14
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:15
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:16
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:17
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:18
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:19
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:20
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:21
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:22
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:23
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:24
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:25
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:26
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:27
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:28
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:29
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:30
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:31
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:32
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:33
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:34
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:35
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:36
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:37
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:38
Realtek Bluetooth ERROR: patch timerout, retry:

Realtek Bluetooth :3-wire download patch re send:39
Realtek Bluetooth ERROR: H5 patch timed out

liaoqingmo, 2019-03-03:
You can try to turn off hardware flow control in your code.
I solved this problem in this  way@xuxianyin5490 
change file "hciattach_rtk.c"
'''
2269 // mo
2270 SET_FLOW_CONTRL:
2271 /*
2272     if (rtk_hw_cfg.hw_flow_control) {
2273         RS_DBG("hw flow control enable");
2274         ti->c_cflag |= CRTSCTS;
2275 
2276         if (tcsetattr(fd, TCSANOW, ti) < 0) {
2277             RS_ERR("Can't set port settings");
2278             return -1;
2279         }
2280     }
2281     else {
2282 */
2283         RS_DBG("hw flow control disable");
2284         ti->c_cflag &= ~CRTSCTS;
2285   //  }
2286 '''

ruschigo, 2019-09-27:
Hi!

I solved my issue removing the config file. 
sudo rm /lib/firmware/rtl_bt/rtlbt_config

then it works for me, even getting some errors at the startup.

Now i have it working with 1500000 of baudrate.